GwhsRobotics3/Team-5507-2018 #1: Consider whether XBox rumble functionality would be useful

gregmarra, 2018-01-22:
We can pass feedback to the operator using the XBox controller rumble feature.

See example here: https://github.com/Team319/frc319-2018/blob/master/src/org/usfirst/frc/team319/models/BobController.java

GwhsRobotics3, 2018-02-01:
GREG HELP! We are trying to do rumble and we are wondering if we should make a subsystem for joystick since rumble does not really belong to any part of the robot. Also, do commands need a subsystem dependency? We tested the limit switches successfully and can you look at our SmartElevator code and give us some tips on how to start the setDesiredPosition method? Thanks

gregmarra, 2018-02-01:
Checking this out now :)

gregmarra, 2018-02-01:
## Should we make a subsystem for the XBox controller?

We could make a new Class that represents the XBox controller, but it probably shouldn't be a Subsystem. Subsystems extend the Subsystem class, which comes with a bunch of functionality we won't need for the XBox controller. Instead, we can extend the Joystick class.

In the same way we are created "RobotMap" as a class in RobotMap.java, we could create a class called XBoxController, and have it extend the wpilib Joystick class. Then anywhere we were previously creating and passing Joystick objects around, we can create and pass around XBoxController objects. It will have all the same methods Joystick has, plus any new ones we create (like `rumble()`)

``` 
public class BobController extends Joystick {
  ...

  public void rumble() {
    ...
  }
}
```

gregmarra, 2018-02-01:
## Do commands need a subsystem dependency?

If a command doesn't require any subsystem, I do not believe it would have any subsystem dependency. The subsystem dependencies are to make sure two different Commands don't both try to drive the same system in opposite directions at the same time.

You're thinking about things like test Commands to print things or rumble the XBox controller?

gregmarra, 2018-02-01:
## SmartElevator
See comments in #7. I'm going to keep comments related to the Issue they're about :)

GwhsRobotics3, 2018-02-02:
Hi Greg, we made a new xboxcontroller class and replaced the joystick object with genericHID. We have not tested what this effect will have on the rest of our code but we changed OI to work with our xboxcontroller class. 
Hopefully it will work well,
Software Team

gregmarra, 2018-02-02:
Were you able to test that the controller can command the robot and that the rumbling works?

gregmarra, 2018-02-02:
I'm doing a bit more reading, and it looks like WPILib may have done some of the heavy lifting for us. Let's try to use their implementation of XBoxController instead of building our own.

1. They have an XBoxController class: http://first.wpi.edu/FRC/roborio/beta/docs/java/edu/wpi/first/wpilibj/XboxController.html
  1. Because it extends edu.wpi.first.wpilibj.GenericHID , we get the method setRumble() on it already.
  1. We probably do not need our own XBoxController class, and instead can delete that file and import XBoxController from wpilib in OI.java. We should correct the imports in OI.java, but after we delete the file because we had an identical class name we won't even need to modify https://github.com/GwhsRobotics3/Team-5507-2018/blob/master/src/org/usfirst/frc/team5507/robot/OI.java#L63
1. The JoystickButton class (which we use to send the Commands from OI.java) will work with the WPILib XBoxController class
  1. It might seem like the JoystickButton class needs to be passed a Joystick, but if you look at the documentation we can see it just needs to be passed a GenericHIDController: http://first.wpi.edu/FRC/roborio/beta/docs/java/edu/wpi/first/wpilibj/buttons/JoystickButton.html .. XBoxController extends GenericHIDController, so we're all set. (HID stands for Human Input Device).

GwhsRobotics3, 2018-02-03:
We got the rumble function to work when we press a button. It's uploaded to github. Next we will test rumble with the limit switches.

gregmarra, 2018-02-03:
Nice! Do you have to set the rumble back to 0 in the command end() method to get it to stop rumbling?

Also, make sure you check out my Piazza post with an example of one way to do SmartGripper.

GwhsRobotics3, 2018-02-03:
Yeah we do. Sometimes it rumbles even when we disable the robot.

gregmarra, 2018-02-03:
How could you make it stop rumbling after a bit?

GwhsRobotics3, 2018-02-06:
we added a timer

closed 2018-02-08 by GwhsRobotics3