GwhsRobotics3/Team-5507-2018 #7: [Elevator] Make 'SmartElevator' subsystem

gregmarra, 2018-01-31:
We can create duplicate subsystems, one version "Smart" that uses closed loop control based on encoders and other sensors, and a version "Dumb" that doesn't.

# Two Classes

For example, we can make a Smart and Dumb elevator:

- DumbElevator: Only has a "go up" and "go down" command that drives the elevator as long as a button is being held.
- SmartElevator: Has a "go up one position" and "go down one position" command, keeps track of its own current position, and uses an encoder to decide when its reached the new desired positions.

Since we want the robot to be able to work and tested even before we have finished all the mechanical systems, this will let us quickly swap out the code by changing one line:

In Robot.java, we can change
`public static Elevator m_elevator = new Elevator();`
to instead be
`public static SmartElevator m_elevator = new SmartElevator();`

## Keeping track of states

SmartElevator can have its own internal state for what position it was last in.

This not-quite-java illustrates

```
Inside of the Subsystem, we can have some variables to track the state

function getNextStateDown() {
  if (current_state == "low") { return "low" }
  if (current_state == "medium") { return "low" }
  if (current_state == "high") { return "medium" }
}

function getPositionForState(state) {
  if (state == "low") { return RobotMap.ElevatorLowPosition }
  ...
}
```

And then in the "goDownCommand", we'd say

```
execute() {
  elevator.setDesiredPosition(getPositionForState(getNextStateDown())
}
```

# Mechanical Mockup

We can also build a mock-up of the SmartElevator mechanical system using our encoder test rig. We can hook that test rig motor up to our Talon that will drive the SmartElevator (or any Talon, if we remap things in RobotMap.java), and then demonstrate that the system works as expected.

Can we get the mechanical mockup to toggle between "low", "medium" and "high" positions?

## Unknown Constants

We won't know yet what position values will correspond to "low", "medium" and "high" for the elevator, so we can make constants for now in Constants.java and reference them in the code. Then when we have the real robot built, we can move the elevator to the right positions, note down the positions, and change our constants, and the code will be all ready.

gregmarra, 2018-01-31:
(If we have encoders added to the grabber arms first, we may want to do a SmartGrabber first)

gregmarra, 2018-02-01:
# Some pointers!

The start you have for SmartElevator looks nice! Here are a few questions and pointers.

I'm looking at https://github.com/GwhsRobotics3/Team-5507-2018/blob/master/src/org/usfirst/frc/team5507/robot/subsystems/SmartElevator.java

## [hi-pri] The Robot should only have one "Elevator" at a time.
Right now, you're creating both a Elevator and SmartElevator for the robot.

```
public static Elevator m_elevator = new Elevator();
public static SmartElevator m_smartElevator = new SmartElevator();
```

I think the robot is going to get unhappy having two different WPI_TalonSRX objects that are created for the same CAN ID. What would happen if you told the `Elevator` to go up and the `SmartElevator` to go down at the same time?

I think that at any time, you should set `Robot.m_elevator` to be _either_ an Elevator _or_ a SmartElevator. This will prevent this case of creating two separate WPI_TalonSRX objects that refer to the same physical motor controller, since you will only create them in either an Elevator or a SmartElevator.

## [mid-pri] getPositionForState() should take a state as an argument

Rather than using the robot's current state, allow looking up the position for _any_ state, and have callers of this method pass in the desired state as an argument.

This would allow calls like:
```
m_elevator.getPositionForState(m_elevator.getNextStateUp())
```

I'll add more notes about how to track the SmartElevator's state internally in another comment.

## [low-pri] Be explicit with Constant names.

It's excellent that you've set the Elevator heights in Constants! This will make them fast and easy to adjust.

However, the names don't indicate that they are for the Elevator system. What if we end up with a "highPosition" value for the Climber as well?

https://github.com/GwhsRobotics3/Team-5507-2018/blob/master/src/org/usfirst/frc/team5507/robot/Constants.java#L17

You may want to name them elevatorHighPosition, etc.

Alternatively, if we think we will only ever refer to these in the SmartElevator class, we could make them properties of the SmartElevator class. Then you'd refer to them as `SmartElevator.HIGH_POSITION`. (All-caps is often used to denote constants). (Also, I recognize that if you move these back into SmartElevator it'd be undoing my recommendation to put them in Constants.java, but it might actually be a bit cleaner to keep them in SmartElevator.)

gregmarra, 2018-02-01:
## Naming Conventions
* getXYZ - "get" as a prefix usually means you are "reading something, and not changing anything in the process. For example, in `getPositionForState`, you are reading a number, perfect!
* setXYZ - "set" as a prefix usually means you are "changing something". For example, your current function `getNextStateUp` actually _changes_ SmartElevator.currentState to a different value! You should rename this method to `setNextStateUp` to be clear it changes something. "Changing Something" is sometimes called "having a side effect".

## Setting Position
The SmartElevator should use the WPI_TalonSRX "magic motion" similarly to the way we made the Subsystem for our Encoder Demo in `EncoderDemo.driveToAngle`. Think about the similarities between the mechanical system that the EncoderDemo and the SmartElevator are driving. How are they similar? How are they different?

GwhsRobotics3, 2018-02-02:
We tried coding the elevator going up and down based on the other methods but it is not working. Is there something wrong with the logic of the code?

gregmarra, 2018-02-02:
## Using Counters

It looks like you're basing the counter code on https://wpilib.screenstepslive.com/s/3120/m/7912/l/599687-using-limit-switches-to-control-behavior .

I think that this is a bit more complicated than it needs to be for now. Let's be simpler, and drop the Counters, and just based things on "Command-based program to operate until limit switch closed"

1. Remove the Counters from SmartElevator.
2. Change isSwitchSetTop and isSwitchSetBottom to return the DigitalInput's value directly via it's get() method: http://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/DigitalInput.html#get--

## Don't use while loops

We will almost never want to use while loops in our robot code. We already are calling methods over and over inside of a larger loop (the execute() method of Commands gets called continuously while the command is running, so we don't need to use a while loop to make sure something keeps happening).

Right now you have a while here: https://github.com/GwhsRobotics3/Team-5507-2018/blob/master/src/org/usfirst/frc/team5507/robot/subsystems/SmartElevator.java#L97 . You can change that while to an if, and it will run over and over thanks to the Command calling execute() repeatedly.

## What will happen when getCurrentPos() > pos?
After you change https://github.com/GwhsRobotics3/Team-5507-2018/blob/master/src/org/usfirst/frc/team5507/robot/subsystems/SmartElevator.java#L97 to be an if, what will happen once the SmartElevator has lifted above the current set position, and getCurrentPos() exceeds pos?

gregmarra, 2018-02-02:
## Using the Talon's closed loop control.

The biggest thing we're going to want to change in SmartElevator is to use the Talon's built in Position Control.

Think about EncoderDemo. What's different between what EncoderDemo does, and what we want our elevator system to do?

closed 2018-02-08 by gregmarra